ros/urdfdom #58: groovy version

goretkin, 2015-03-10:
From what I can understand from the discussion in https://github.com/ros/urdfdom/issues/1, urdfdom is/was a package in groovy, but I do not see a git branch for it. Also there is a file at  https://raw.githubusercontent.com/ros-gbp/urdfdom-release/debian/groovy/precise/urdfdom/package.xml

I do see https://github.com/eacousineau/urdf_parser_py, which mentions groovy in the repo description. And I do see https://github.com/laas/robot_model_py/tree/groovy which is not a package called "urdfdom", but it seems this repo and that repo share a common history.

I didn't check all of the history, but my guess is that the project was copied over at this commit
https://github.com/ros/urdfdom/commit/ba8d4de4cd811374e490bb93c825677f7590eb06
https://github.com/laas/robot_model_py/commit/ba8d4de4cd811374e490bb93c825677f7590eb06

How can we restore the groovy branch (and possibly older versions of ROS, but I'm on groovy) into this repo?

If there were a groovy-devel branch, I would, for example, apply https://github.com/ros/urdfdom/pull/38 to it.

goretkin, 2015-03-10:
I don't know how this repo and `robot_model_py` came to be. Perhaps they were both transfered from the same repo in bitbucket.

This tool https://metacpan.org/pod/git-stitch-repo might

jacquelinekay, 2015-12-09:
hey @goretkin!!!

It's a bit hidden, but what about the `robot_model-1.6` branch?

https://github.com/ros/urdfdom/tree/robot_model-1.6/urdf_parser

I suspect this this is the state that the package currently known as urdfdom was in when Groovy was released.

I think this repo was formerly robot_model and was renamed urdfdom and stripped of most of its contents (which was moved into the current robot_model repo) in hydro.

And I believe @eacousineau's urdf_parser_py was copied into `urdf_parser_py` in this repo.

I wasn't at Willow Garage or anything but I think this is the story.

Groovy is EOL'd so I'm not inclined to spend a lot of time maintaining a groovy-devel branch, but I could start one (branching off of robot_model-1.6?) if you would like to apply patches to it.

goretkin, 2015-12-10:
hi @jacquelinekay! Thanks for digging into it. I think it would only be helpful to have a groovy-devel branch if enough people are still using Groovy. I know it's true for most PR2 users. I've been relying on my own version of urdfdom that I've used. I haven't touched it in a while because I didn't run into any other problems.

sloretz, 2017-08-10:
It looks like it's been a while since this issue was last updated, and there hasn't been much demand for a groovy-devel branch. I will close this issue, but feel free to reopen if you would like to see progress on it.

closed 2017-08-10 by sloretz