ros/urdfdom #62: Use c++11 and remove boost dependency

scpeters, 2015-08-01:
Are there any plans to make a version of urdfdom that uses c++11 and replaces `boost::*_ptr` with `std::*_ptr`? One semi-migration strategy could be to start using typedefs:

~~~
typedef boost::shared_ptr<urdf::Link> UrdfLinkPtr;
~~~

and then in the next major version of urdfdom switch them over to `std::` instead of `boost::`

~~~
typedef std::shared_ptr<urdf::Link> UrdfLinkPtr;
~~~

Then anyone who wants their code to support both versions could use the typedefs? I'm sure there's big holes in this strategy (such as `boost::*_pointer_cast` vs `std::*_pointer_cast`), but you've got to start somewhere.

scpeters, 2015-08-02:
@wjwwood any thoughts?

wjwwood, 2015-08-02:
We don't have a plan for migrating to C++11 types within ROS 1, but adding a typedef which doesn't already exist which points to the C++11 types is probably a good strategy. I'm not sure what the sticking points will be in the transition, but if I see a concrete proposal (pull request) I can give feedback on what might be affected based on that.

We already guarantee that ROS 1 will compile when `-std=c++11` is passed to the compiler, so we only have to consider the upgrade with respect to types that have changed. We probably won't be able to drop the boost dependency anytime soon. Because we use `boost::filesystem` and `boost::program_options` and a few other things which have no C++11 equivalent.

Quickly looking at `urdfdom` it uses `boost::lexical_cast`:

https://github.com/ros/urdfdom/blob/ccc7d2ff6653e190907e5023379b757784a36cea/urdf_parser/src/urdf_model_state.cpp#L41

Which is probably replaced by `std::to_string`.

It also uses `shared_ptr` in the public API:

https://github.com/ros/urdfdom/blob/ccc7d2ff6653e190907e5023379b757784a36cea/urdf_parser/include/urdf_parser/urdf_parser.h#L61
https://github.com/ros/urdfdom_headers/blob/56471a55d85e1be27a91a5f2f72b001d92cddb15/urdf_sensor/include/urdf_sensor/sensor.h#L157
(and others)

And `weak_ptr`:

https://github.com/ros/urdfdom_headers/blob/56471a55d85e1be27a91a5f2f72b001d92cddb15/urdf_model/include/urdf_model/link.h#L239

isucan, 2015-08-03:
my concern is that there is a lot of code that depends on these types, so I am not too inclined to change this. What do you think of forking to an urdfdom2 that uses c++11 and avoids boost?

scpeters, 2015-08-05:
Adding c++11 is a huge change, so I had assumed that would involve a major version bump. I had imagined that it could just live in a new branch. Do we need to fork to make a package with a new name (like `urdfdom2`) for this type of change?

scpeters, 2015-09-12:
Bump my previous comment: can we make breaking changes in a new release branch (bumping the major version), or does it need to be in a new repository?

wjwwood, 2015-09-14:
I'd say do it as a branch, but @isucan should make the call.

isucan, 2015-09-14:
Please do this in a new branch, no need for a new repo.

scpeters, 2015-09-15:
I just made a pull request to urdfdom_headers to create typedefs for smart pointers. It should preserve backwards compatibility but make it easier to migrate away from boost.

https://github.com/ros/urdfdom_headers/pull/13

scpeters, 2016-07-20:
boost is gone!

closed 2016-07-20 by scpeters