ros/urdfdom #66: boost::reinterpret_pointer_cast issue in types.h

binary42, 2015-10-02:
in types.h for /usr/local/include/urdf_model, the function reinterpret_pointer_cast should return urdf:: not boost:: for successful compilation

scpeters, 2015-10-02:
Can you elaborate? The code compiles find on my computer. Are you observing a build failure? If so, please give more details.

binary42, 2015-10-02:
Absolutely, sorry, I was in the middle of three things at the time of the post. I am not sure of the build order and when /usr/local/include/urdf_model/ files get built, but after they are created, during make install, I get a build failure, stating that boost::reinterpret_pointer_cast in types.h should be urdf::reinterpret_pointer_cast, i.e. the method reinterpret_pointer_cast, which returns boost::reinterpret_pointer_cast shout return urdf::reinterpret_pointer_cast.

As to the deeper reason, I haven't looked further into it. Once I modify types.h to return the urdf scoped reinterpret_pointer_cast, the make install compiles with no issue. Also, I am compiling on a raspberry pi b+ ... so that in itself is a bit esoteric. Following install of ROS for raspberry pi from wiki.ros.org/ROSberryPi

scpeters, 2015-10-02:
Can you post a patch?

scpeters, 2015-10-03:
Can you try including `boost/pointer_cast.hpp.` in urdfdom_headers?

~~~
diff --git a/urdf_model/include/urdf_model/types.h b/urdf_model/include/urdf_model/types.h
index 5b2b5e6..4c761e2 100644
--- a/urdf_model/include/urdf_model/types.h
+++ b/urdf_model/include/urdf_model/types.h
@@ -37,6 +37,7 @@
 #ifndef URDF_MODEL_TYPES_H
 #define URDF_MODEL_TYPES_H
 
+#include <boost/pointer_cast.hpp.>
 #include <boost/shared_ptr.hpp>
 #include <boost/weak_ptr.hpp>
 
~~~

binary42, 2015-10-09:
@scpeter, Hi Steven, yes, the include of boost/pointer_cast.hpp also fixes the issue, and seems more appropriate. Thanks.

scpeters, 2015-10-09:
Ok, I've added the fix to https://github.com/ros/urdfdom_headers/pull/17

isucan, 2015-10-11:
This has now been merged, should the issue be considered fixed?

scpeters, 2015-10-12:
I think so, we can let @binary42 reopen if it persists.

scpeters, 2015-11-09:
Yes, this can be closed.

closed 2015-12-09 by jacquelinekay