ros/urdfdom #78: Actuator inertia

nielsvd, 2016-02-11:
Actuators used in industrial robotic arms typically have a reasonably low inertia, which can be included in the link inertia. However, the actuator inertia has another important effect on the rigid body dynamics; namely, due to the gear ratio between the motor and the joint, the actuator inertia significantly contributes to the required torque to achieve a certain requested motion.

At the moment it seems that the URDF specification does not support the specification of such contribution of motor inertia. I think it could be well-placed in the [<transmission>](http://wiki.ros.org/urdf/XML/Transmission) tag. 

Some relevant questions for the discussion:
- Am I missing out on something (actuator inertia is supported)?
- If not, is it in accordance to the philosophy behind URDF to include actuator inertia in the specification?
- If so, do you agree it should be place in the <transmission> element?

Thanks in advance for your input.

jacquelinekay, 2016-02-17:
Correct, neither the transmission element nor the actuator element have inertia in URDF.

If we were to introduce actuator inertia, it would probably be as a sub-element of the actuator element, which is a sub-element of transmission.

I'm not a mechanical engineer, so can you provide a specific example, perhaps a formula for a motor model, where actuator inertia is needed and cannot be faked by some combinator of the joint dynamics and limit sub-elements?

nielsvd, 2016-03-02:
Sorry for my late reply. Please find an example below (it's included as an image to display the maths nicely).

[image]

scpeters, 2016-03-02:
This is excellent documentation! Thanks for this example. Here's my comments:

* I assume that `v_k` is measured at the link center-of-mass.

* As you suggest in the issue description, I think the `transmission` tag would be a good place for the actuator inertia.

* Is there any case when the actuator inertia would not be a scalar, assuming it acts along a single joint axis?

nielsvd, 2016-03-03:
In response to the comments by @scpeters:
 * Yes `v_k` is measured in the link center-of-mass, an important note I forgot to state in the example.
 * I am not aware of a case where a scalar-valued inertia doesn't suffice, indeed assuming the actuator acts along a single joint axis.

neemoh, 2020-04-14:
Any update on this? I am surprised that motor inertia is not included in URDF. It is not something one can ignore when dealing with dynamics. Just to give you an example, the motor inertias (considering the gear ratio) of the first 2 joints of a 5kg payload robot can contribute 30-40% to the diagonals of the mass matrix.

emekBaris, 2021-04-02:
Is there any update?