ros/urdfdom #82: sensor parsing

rhaschke, 2016-03-09:
urdfdom has code to parse `<sensor>` tags according to this [spec](http://wiki.ros.org/urdf/XML/sensor).

However, this code doesn't seem to be used. It goes to liburdfdom_sensor, but as [symbols are hidden by default](https://github.com/ros/urdfdom/blob/fb023fe/CMakeLists.txt#L26-L28), it cannot be used externally.

We are currently in the process of proposing a new [Tactile sensor](https://github.com/ubi-agni/urdfdom/tree/tactile_sensors) and would like to add the list of parsed sensors to the ModelInterface (see [here](https://github.com/ubi-agni/urdfdom_headers/commit/f9995d36434ffd6bdb8e7f2294ace32f0fcf8103) for our proposal). Any objections to that strategy? Should the robot model be kept as lean as possible instead, leaving the sensor parsing to some separate function returning a list of sensors?

Please comment on the chosen approach and we will adjust accordingly and eventually file a PR.

rhaschke, 2016-03-13:
Thinking deeper on the topic and considering discussion in #28, I now believe, that we need a plugin mechanism to handle the infinite number of available sensors out there. I will further think about that and make a proposal.

closed 2016-03-13 by rhaschke

jacquelinekay, 2016-03-13:
Thanks @rhascke, I was composing a response pointing to that discussion but you beat me to it. :)

The discussion for adding sensors to URDF has surfaced and resurfaced a few times. As you suggested, the difficulty of supporting sensors is choosing the right abstractions.

SDF supports sensors because Gazebo can simulate a specific set of sensors (laser range finders, cameras, etc.).

One question I have is how sensors in URDF will be integrated with the rest of the ROS toolchain. For example, when a robot with a LaserScan <sensor> tag in its URDF is visualized in Rviz, will Rviz know to add a LaserScan visualization with the correct transform extracted from the URDF? These kinds of questions might help motivate the proposal and make it general enough to benefit the whole community.

I'm open to reviewing new proposals for sensors in URDF. However we should consider the development overhead of the initial implementation as well as the integration with our tooling. A bit of time and effort investment in the design may help reduce that overhead.

rhaschke, 2016-03-14:
> One question I have is how sensors in URDF will be integrated with the 
> rest of the ROS toolchain. For example, when a robot with a LaserScan 
> tag in its URDF is visualized in Rviz, will Rviz know to add a 
> LaserScan visualization with the correct transform extracted from the 
> URDF? These kinds of questions might help motivate the proposal and 
> make it general enough to benefit the whole community.
>
Currently, we are not (yet) thinking about an automated creation of 
additional displays in rviz. But our work points along a similar direction:
Given you add a sensor stream visualization to rviz (in our current case 
tactile data), the associated display plugin will inspect the urdf to 
know about sensor transform and other sensor parameters, which are fixed 
and therefore are not published in the sensor stream.

> I'm open to reviewing new proposals for sensors in URDF. However we 
> should consider the development overhead of the initial implementation 
> as well as the integration with our tooling. A bit of time and effort 
> investment in the design may help reduce that overhead.
>
Thanks, I will keep you updated ;-)

kavonszadkowski, 2016-03-17:
I'm less of a ROS insider and come from a simulation-biased perspective, but I second Robert's notion of introducing a plugin mechanism. In [MARS](https://github.com/rock-simulation/mars/) we annotate URDF with information about sensors using [SMURF](https://github.com/rock-simulation/smurf_parser) and use factories to create them in simulation.

But maybe that's something more fundamental that may apply to other elements beyond sensors and thus should be discussed along with other improvements for "URDF 2.0"?

jacquelinekay, 2016-03-18:
I have a lot of ideas about extensibility/a plugin interface for the "next generation" of RDFs. I am writing them up and will post them on a public forum (and notify you both when the post is finished).

traversaro, 2016-03-18:
@rhaschke some times ago we set up a ROS SIG on Robotic Skins [1]. 

While activity on the SIG has been relativly low, I think you could find useful feedback on the semantics of your proposal on how to describe tactile sensors information, even before that discussion on how to include sensors information on the "next generation of *RDFs" is completed. 

[1] :  http://wiki.ros.org/sig/RobotSkin

kavonszadkowski, 2016-03-18:
Sachin Chitta has put together some notes on the topic of future RDFs and has pulled a number of my suggestions, again coming from SMURF. I haven't yet found time to work in the parallels between SDF and SMURF, but a number of ideas are in there.

It's on a [github page](http://sachinchitta.github.io/urdf2/) rendered from [this repository](https://github.com/sachinchitta/urdf2).

Perhaps we could move that repo into **ros** and use it as a starting point for some kind of white paper or alternatively use a github wiki to put our notes together?

jacquelinekay, 2016-03-18:
Some thoughts here:
http://discourse.ros.org/t/urdf-ng-parser-tools-wishlist/55

We would need @sachinchitta's permission to move the repo into the ROS org and I wouldn't want to fork it into the organization without asking him first :) Sachin, do you currently have any interest in helping to push major changes to URDF/whatever robot description format is used in ROS? It would be great to have your input.