ros/urdfdom #91: Add appveyor windows CI

scpeters, 2016-07-20:
We currently have travis CI for linux, but I'd also like to have Windows CI with appveyor. The install script shouldn't be too hard to make, but as I took a first look at it, I realized we would have to install console_bridge and urdfdom_headers. It would be nice if we could package those for chocolatey. Does anyone have experience with packaging for chocolatey? @traversaro @jslee02 ?

traversaro, 2016-07-20:
I never had experience in packaging for Chocolatey. As far as I understood Chocolatey was just for programs, while NuGet was meant to install libraries, but was not supporting global installation of libraries. 
Anyhow I remember that I saw something similar (a upstream project publishing binaries for Windows for use in the CI in a downstream project) in the AppVeyor configuration of simbody and opensim: 
https://github.com/simbody/simbody/blob/master/appveyor.yml#L40
https://github.com/opensim-org/opensim-core/blob/master/appveyor.yml#L64

Anyhow in my experience for such small libraries (such as `console_bridge` and `urdfdom_headers`) the easiest solution is to build everything in AppVeyor. To simplify this process I usually use a superbuild based on CMake's ExternalProject [1], but I guess that using something like catkin or ament (if they support Windows nowadays) would be equivalent. 

[1] : http://robotology.github.io/ycm/gh-pages/master/index.html#superbuild

scpeters, 2016-07-20:
Thanks for the tips! I'll probably start with building both from source, and then look into the nuget or superbuild references.