ros/urdfdom #92: Create the release 1.0.0

j-rivero, 2016-07-21:
I believe that after the merge of pull request #62  we are ready to tag and release the version 1.0.0.  
If there is anything missing, please let me know.  

//cc @scpeters

scpeters, 2016-07-21:
There are some outstanding pull requests, but they are proposing new features, which we still need to figure out how to handle. I think we can go with 1.0

j-rivero, 2016-07-21:
Agree. I don't have perms to create the release, could you please generate it?

scpeters, 2016-07-21:
draft:

* https://github.com/ros/urdfdom/releases/tag/untagged-9211ae09e91a28ad74ce

scpeters, 2016-07-21:
I also created a `0.4` branch which is two pull requests ahead of 0.4.2

* #88 
* #89

scpeters, 2016-07-21:
https://github.com/ros/urdfdom/releases/tag/untagged-1789fe5a1ae3ecd8f919

j-rivero, 2016-07-22:
> https://github.com/ros/urdfdom/releases/tag/untagged-1789fe5a1ae3ecd8f919

Probably reserve for repo admins only, 404 for me.

scpeters, 2016-07-22:
https://github.com/ros/urdfdom/releases/tag/1.0.0

closed 2016-07-22 by scpeters