ros/urdfdom #98: urdfdom usage of std::stod is error-prone to changes in system locale

traversaro, 2017-05-05:
The use in several parts of urdfdom of the function `std::stod` can be source of errors if the library is used together with other libraries (such as Qt) that change the system locale. 

Slightly related, I assume (from all the ROS code using the library) the correct way of encoding floating point numbers as strings in the URDF file format is the one defined by the `"C"` standard locale, but this is not actually documented anywhere (see https://github.com/robotology/idyntree/issues/288#issuecomment-299447873). 

Downstream issue: https://github.com/robotology/idyntree/issues/288
Similar SDFormat issue: https://bitbucket.org/osrf/sdformat/issues/60

traversaro, 2017-05-05:
This is not directly related to `urdfdom`, but apparently a lot of custom URDF/SDF parser available in existing libraries suffer the same problem. 

I just checked and I easily found several examples:
* DART https://github.com/dartsim/dart/blob/52bbd15f8d01e97ed4bdf91b7731daffb4929de7/dart/utils/XmlHelpers.cpp#L167 (lexical_cast is also affected, http://stackoverflow.com/questions/18980423/locale-invariant-guarantee-of-boostlexical-cast , cc @jslee02 )
* Bullet  https://github.com/bulletphysics/bullet3/blob/79d24db60f2c916ffcdaf8b074be622e2e5bc609/examples/Importers/ImportURDFDemo/urdfLexicalCast.h#L7
* Drake https://github.com/RobotLocomotion/drake/blob/4532d9460627535e5feb7d0f19ce951b35b19f95/drake/multibody/parsers/xml_util.cc#L70
* Simbody https://github.com/simbody/simbody/blob/3469006e99a94e9a3c7201a5308d298d20acf07f/SimTKcommon/include/SimTKcommon/internal/String.h#L395

traversaro, 2017-05-05:
cc @liesrock this is the issue I was mentioning today that is also affecting RBDL indirectly.

liesrock, 2017-05-09:
thanks @traversaro, I cc @alaurenzi that can be also interested.

traversaro, 2018-02-21:
Fixed by https://github.com/ros/urdfdom/pull/105 .

closed 2018-02-21 by traversaro