ros/urdfdom #108: Add attribute to indicate if a mesh is a convex hull?

Levi-Armstrong, 2018-01-23:
Would the maintainer be open to adding an an attribute to the mesh tag indicating if the mesh is a convex hull? The default would be false.

`<mesh  filename="" convex="true" />`

sloretz, 2018-04-13:
Hi @Levi-Armstrong, did you have a particular use-case? It might be a topic worth opening up on https://discourse.ros.org/c/robot-description-formats

Levi-Armstrong, 2018-04-16:
@sloretz As @gavanderhoorn mentioned this is the main factor. Most of the collision libraries use libccd to perform convex-convex collision checking and it also provides the minimum translation vector when in collision which would could be used by several optimization based motion planners.

de-vri-es, 2018-04-16:
> Hi @Levi-Armstrong, did you have a particular use-case? It might be a topic worth opening up on https://discourse.ros.org/c/robot-description-formats

Thread on discourse: https://discourse.ros.org/t/add-attribute-to-indicate-if-a-mesh-is-a-convex-hull/4479/2

briansoe66, 2025-01-17:
@Levi-Armstrong Tesseract URDF Package defines a urdf version 2 with extra elements, including a <convex_mesh> to use in place of <mesh>. But this new urdf version does not seem to be very popular, because i cannot find many urdf 2 examples for popular robots. I prefer that urdfdom adds a <mesh convex="true"> attribute like you suggested.

https://ros-industrial-tesseract.readthedocs.io/en/stable/_source/tesseract_urdf_doc.html

Levi-Armstrong, 2025-01-17:
The reason you probably do not find any reference is because the default behavior of tesseract is to convert all meshes to convex hulls.