ros/urdfdom #117: Please respect the standard cmake variable BUILD_TESTING

yurivict, 2018-11-20:
Please do not build unit tests unless this variable is set.

See here: https://cmake.org/cmake/help/v3.0/module/CTest.html

Thanks,
Yuri

traversaro, 2018-11-20:
I guess this was solved by https://github.com/ros/urdfdom/pull/111 .

yurivict, 2018-11-20:
No, the problem exists in 1.0.0-9-g1857a55.

traversaro, 2018-11-20:
I can't find the commit you reference in this repo ( https://github.com/ros/urdfdom/commit/g1857a55 ). 
You are referring to a specific version in some package or port repository?

yurivict, 2018-11-20:
1.0.0-9-g1857a55 is git id, the corresponding commit is 1857a55

scpeters, 2018-11-29:
I just tested this an it appears to be working. Which unit tests are getting built for you with `-DBUILD_TESTING=OFF`?