ros/urdfdom #118: Tests fail

yurivict, 2018-11-20:
```
ninja: no work to do.
[0/1] cd /usr/ports/misc/urdfdom/work/.build && /usr/local/bin/ctest --force-new-ctest-process
Test project /usr/ports/misc/urdfdom/work/.build
    Start 1: urdf_double_convert
1/4 Test #1: urdf_double_convert ..............   Passed    0.00 sec
    Start 2: urdf_double_convert_locale
2/4 Test #2: urdf_double_convert_locale .......   Passed    0.00 sec
    Start 3: urdf_unit_test
3/4 Test #3: urdf_unit_test ...................***Exception: SegFault  0.00 sec
    Start 4: urdf_unit_test_locale
4/4 Test #4: urdf_unit_test_locale ............***Exception: SegFault  0.01 sec

50% tests passed, 2 tests failed out of 4

Total Test time (real) =   0.02 sec

The following tests FAILED:
	  3 - urdf_unit_test (SEGFAULT)
	  4 - urdf_unit_test_locale (SEGFAULT)
```

FreeBSD 11.2 amd64

traversaro, 2018-11-20:
Can you run `ctest` with the `-VV` option to obtain a more verbose output? Thanks!

yurivict, 2018-11-20:
```
$ ctest -VV
UpdateCTestConfiguration  from :/usr/ports/misc/urdfdom/work/.build/DartConfiguration.tcl
Parse Config file:/usr/ports/misc/urdfdom/work/.build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/usr/ports/misc/urdfdom/work/.build/DartConfiguration.tcl
Parse Config file:/usr/ports/misc/urdfdom/work/.build/DartConfiguration.tcl
Test project /usr/ports/misc/urdfdom/work/.build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: urdf_double_convert

1: Test command: /usr/ports/misc/urdfdom/work/.build/bin/urdf_double_convert "--gtest_output=xml:/usr/ports/misc/urdfdom/work/.build/test_results/urdf_double_convert.xml"
1: Environment variables: 
1:  LC_ALL=C
1: Test timeout computed to be: 240
1: Running main() from gtest_main.cc
1: [==========] Running 2 tests from 1 test case.
1: [----------] Global test environment set-up.
1: [----------] 2 tests from URDF_DOUBLE_CONVERT
1: [ RUN      ] URDF_DOUBLE_CONVERT.test_successful_conversion
1: [       OK ] URDF_DOUBLE_CONVERT.test_successful_conversion (0 ms)
1: [ RUN      ] URDF_DOUBLE_CONVERT.test_failed_conversion
1: [       OK ] URDF_DOUBLE_CONVERT.test_failed_conversion (0 ms)
1: [----------] 2 tests from URDF_DOUBLE_CONVERT (0 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 2 tests from 1 test case ran. (0 ms total)
1: [  PASSED  ] 2 tests.
1/4 Test #1: urdf_double_convert ..............   Passed    0.00 sec
test 2
    Start 2: urdf_double_convert_locale

2: Test command: /usr/ports/misc/urdfdom/work/.build/bin/urdf_double_convert "--gtest_output=xml:/usr/ports/misc/urdfdom/work/.build/test_results/urdf_double_convert_locale.xml"
2: Environment variables: 
2:  LC_ALL=nl_NL.UTF-8
2: Test timeout computed to be: 240
2: Running main() from gtest_main.cc
2: [==========] Running 2 tests from 1 test case.
2: [----------] Global test environment set-up.
2: [----------] 2 tests from URDF_DOUBLE_CONVERT
2: [ RUN      ] URDF_DOUBLE_CONVERT.test_successful_conversion
2: [       OK ] URDF_DOUBLE_CONVERT.test_successful_conversion (0 ms)
2: [ RUN      ] URDF_DOUBLE_CONVERT.test_failed_conversion
2: [       OK ] URDF_DOUBLE_CONVERT.test_failed_conversion (1 ms)
2: [----------] 2 tests from URDF_DOUBLE_CONVERT (1 ms total)
2: 
2: [----------] Global test environment tear-down
2: [==========] 2 tests from 1 test case ran. (1 ms total)
2: [  PASSED  ] 2 tests.
2/4 Test #2: urdf_double_convert_locale .......   Passed    0.00 sec
test 3
    Start 3: urdf_unit_test

3: Test command: /usr/ports/misc/urdfdom/work/.build/bin/urdf_unit_test "--gtest_output=xml:/usr/ports/misc/urdfdom/work/.build/test_results/urdf_unit_test.xml"
3: Environment variables: 
3:  LC_ALL=C
3: Test timeout computed to be: 240
3: [==========] Running 9 tests from 1 test case.
3: [----------] Global test environment set-up.
3: [----------] 9 tests from URDF_UNIT_TEST
3: [ RUN      ] URDF_UNIT_TEST.test_rotation_get_set_rpy_idempotent
3: 
3: before  x: 0.5 y: -0.5 z: 0.5 w: 0.5  roll: 0 pitch: -1.57079633 yaw: 1.57079633
3: after   x: 0.5 y: -0.5 z: 0.5 w: 0.5  roll: 0 pitch: -1.57079633 yaw: 1.57079633
3: ok      1
[... 38 lines cut ...]
4: [       OK ] URDF_UNIT_TEST.test_rotation_get_set_rpy_idempotent (2 ms)
4: [ RUN      ] URDF_UNIT_TEST.test_vector3_simple
4: [       OK ] URDF_UNIT_TEST.test_vector3_simple (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.test_vector3_bad_string
4: [       OK ] URDF_UNIT_TEST.test_vector3_bad_string (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.test_vector3_invalid_number
4: [       OK ] URDF_UNIT_TEST.test_vector3_invalid_number (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.test_vector3_not_enough_numbers
4: [       OK ] URDF_UNIT_TEST.test_vector3_not_enough_numbers (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.test_vector3_too_many_numbers
4: [       OK ] URDF_UNIT_TEST.test_vector3_too_many_numbers (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.parse_joint_doubles
4: [       OK ] URDF_UNIT_TEST.parse_joint_doubles (0 ms)
4: [ RUN      ] URDF_UNIT_TEST.parse_link_doubles
4/4 Test #4: urdf_unit_test_locale ............***Exception: SegFault  0.01 sec

50% tests passed, 2 tests failed out of 4

Total Test time (real) =   0.02 sec

The following tests FAILED:
	  3 - urdf_unit_test (SEGFAULT)
	  4 - urdf_unit_test_locale (SEGFAULT)
Errors while running CTest
```

traversaro, 2018-11-20:
Which version `TinyXML` are you using? Is `TIXML_USE_STL` defined when you compiled TinyXML?

yurivict, 2018-11-20:
It picks the older version ```tinyxml-2.6.2_2```, and not the newer one: ```tinyxml2-6.2.0,1```.
```TIXML_USE_STL``` isn't defined.

closed 2022-02-14 by clalancette