ros/urdfdom #129: Release 1.0.3 into ubuntu bionic

jschleicher, 2019-07-08:
Please release version 1.0.3 also into bionic-updates (currently in [disco](https://packages.ubuntu.com/disco/liburdfdom-dev))

Since this solves the (very popular) issue with RViz https://github.com/ros-visualization/rviz/issues/1249#issuecomment-464503760, also discussed here in #119.

I can confirm, that the rendering in RViz is fixed with a source build of urdfdom_headers version 1.0.3 and urdfdom version 1.0.3 on my melodic installation (Ubuntu bionic).

scpeters, 2019-07-08:
cc @j-rivero

jschleicher, 2019-08-06:
@j-rivero any updates / comments?

j-rivero, 2019-08-06:
To get the update into Bionic we will need to create a SRU issue: https://wiki.ubuntu.com/StableReleaseUpdates . The process needs a good bunch of documentation, I can deal with most of it but it would be good if we can have a test use case (better if it does not need manual inspection, a Dockerfile could work well for this).

gavanderhoorn, 2019-08-06:
@j-rivero: isn't this essentially the same issue as ros/urdfdom#119? That has a `Dockerfile` available that clearly exhibits the problem.

j-rivero, 2019-08-06:
> @j-rivero: isn't this essentially the same issue as #119? That has a `Dockerfile` available that clearly exhibits the problem.

I have no idea if we are speaking about the same bug or a different one. @jschleicher ?

gavanderhoorn, 2019-08-06:
The first post by @jschleicher that started this issue references #119.

j-rivero, 2019-08-06:
> The first post by @jschleicher that started this issue references #119.

you are right, my bad sorry.

gavanderhoorn, 2019-08-06:
So what's the status on #119? :)

jschleicher, 2019-08-09:
Thanks for looking into the release and I'm looking forward for a fixed package.

Closing here as duplicate of #119.

closed 2019-08-09 by jschleicher