ros/urdfdom #137: Build fails with undefined reference to TiXmlElement::SetAttribute

rotu, 2020-04-27:
Ubuntu focal / Clang 11 / libc++
Building the ros2/urdfdom ros2 branch

```
--- stderr: urdfdom                                                                                                                                   
ld.lld: error: ../lib/liburdfdom_model.so.1.0: undefined reference to TiXmlElement::SetAttribute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [urdf_parser/CMakeFiles/check_urdf.dir/build.make:88: bin/check_urdf] Error 1
make[1]: *** [CMakeFiles/Makefile2:294: urdf_parser/CMakeFiles/check_urdf.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
ld.lld: error: ../lib/liburdfdom_model.so.1.0: undefined reference to TiXmlElement::SetAttribute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
ld.lld: error: ../lib/liburdfdom_model.so.1.0: undefined reference to TiXmlElement::SetAttribute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [urdf_parser/CMakeFiles/urdf_mem_test.dir/build.make:87: bin/urdf_mem_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:564: urdf_parser/CMakeFiles/urdf_mem_test.dir/all] Error 2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [urdf_parser/CMakeFiles/urdf_to_graphiz.dir/build.make:87: bin/urdf_to_graphiz] Error 1
make[1]: *** [CMakeFiles/Makefile2:348: urdf_parser/CMakeFiles/urdf_to_graphiz.dir/all] Error 2
ld.lld: error: ../../lib/liburdfdom_model.so.1.0: undefined reference to TiXmlElement::SetAttribute(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [urdf_parser/test/CMakeFiles/urdf_unit_test.dir/build.make:89: bin/urdf_unit_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:1106: urdf_parser/test/CMakeFiles/urdf_unit_test.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< urdfdom	[ Exited with code 2 ]
```

rotu, 2020-04-27:
I do have this package installed:
`libtinyxml-dev/focal,now 2.6.2-4build1 amd64 [installed]`

though CMakeCache shows some weirdness with `TinyXML_DIR`:

```
//The directory containing a CMake configuration file for TinyXML.
TinyXML_DIR:PATH=TinyXML_DIR-NOTFOUND

//Path to a file.
TinyXML_INCLUDE_DIR:PATH=/usr/include

//Path to a library.
TinyXML_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libtinyxml.so
``

rotu, 2020-04-27:
It looks like this is because of `TIXML_USE_STL`, which needs to be set when the library is built. I'm guessing that was not the case when the `libtinyxml-dev` package and its dependencies were built.

I resolved this locally with:
```
colcon build --packages-select tinyxml_vendor urdfdom --cmake-clean-cache --cmake-args -DFORCE_BUILD_VENDOR_PKG=1
```

rotu, 2020-04-27:
@nuclearsandwich is there a compelling reason to keep using `tinyxml` *as well as* `tinyxml2`? (Confusingly, `tinyxml` seems to be on major version number 2) It seems tinyxml is no longer under development and has been superseded by tinyxml2.

clalancette, 2020-04-27:
Unfortunately, there are some downstreams (like urdf, urdfdom, kdl_parser, rviz_common) that still rely on it.  Worse, some of them rely on it in their public headers (which implies a deprecation cycle to deprecate those APIs).  It's a long term goal to get rid of those, but we haven't gotten around to it.

rotu, 2020-04-27:
Created #17 to see if I could. It doesn't seem like anyone's really using these public functions in the ROS2 codebase outside of `urdf` https://index.ros.org/p/urdfdom/github-ros2-urdfdom/#eloquent-deps

traversaro, 2020-04-28:
Given that this is a common source of confusion, it may be worth to mention that actually ROS2 is using a fork of this repo ( https://github.com/ros2/urdfdom ) that unfortunately has quite diverged with respect to the version of `urdfdom` contained in this repo.

clalancette, 2020-04-28:
@traversaro Excellent point.  This looks like one of those repositories where we should merge the two back together, and the update the ros2 branch to look more like the ros1 branch(es).  I'll put it on my TODO list to look at, though this will likely happen after Foxy release.

rotu, 2020-04-28:
:+1: to merging them. The ros2 fork doesn't have an issue tracker, which is why I created the issue here instead.

We could even go a step further and roll the `urdfdom` package into the `urdf` repo.

clalancette, 2020-04-28:
> We could even go a step further and roll the `urdfdom` package into the `urdf` repo.

There are some (additional) complications with that, but I agree in principle.  I'll probably work on merging the ros1 and ros2 forks together first, then we can take up that question.