ros/urdfdom #147: Using wrong headers

audrow, 2020-10-27:
This issue is a continuation of the issue described in https://github.com/ros2/urdfdom/issues/23. Here is its main description:

> When enabling warnings in #22, it seems enabling `-Wpedantic` enabled `-Wgnu-zero-variadic-macro-arguments` in Clang which raised the following warning (as noted here, https://github.com/ros2/urdfdom/pull/22#issuecomment-698637820):
> ```
> [  3%] Building CXX object urdf_parser/CMakeFiles/urdfdom_model.dir/src/pose.cpp.o
> In file included from /Users/osrf/jenkins-agent/workspace/ci_osx/ws/src/ros2/urdfdom/urdf_parser/src/pose.cpp:42:
> /usr/local/include/console_bridge/console.h:67:88: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
>   console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_ERROR, fmt, ##__VA_ARGS__)
> ```
> Two things from this warning: 
> * the console bridge is being included from `usr/local`, and 
> * it is an older version of console_bridge (not `1.0.1`), because the PR https://github.com/ros/console_bridge/pull/64 hasn't been applied yet.

sloretz, 2020-12-29:
@audrow did #148 fix this?

audrow, 2020-12-29:
@sloretz, Yes! Good catch.

PR #148 (2020-11-04): Enable basic warnings + include correct headers + remove Travis CI
This PR is a continuation of https://github.com/ros2/urdfdom/pull/22 and enables `-Wall`, `-Wextra`, `-Wpedantic`. It should also fix #147.
closed 2020-12-29 by audrow