ros/urdfdom #150: Deprecation warning on urdfdom_parser with Clang

Karsten1987, 2021-01-05:
as per output below:
```
/Users/karsten/workspace/ros2/ros2_master/src/ros/urdfdom/urdf_parser/test/gtest/include/gtest/internal/gtest-param-util-generated.h:102:8: warning: definition of implicit copy constructor for 'ValueArray2<bool, bool>' is deprecated because it has a user-declared copy assignment operator [-Wdeprecated-copy]
  void operator=(const ValueArray2& other);
       ^
/Users/karsten/workspace/ros2/ros2_master/src/ros/urdfdom/urdf_parser/test/gtest/include/gtest/gtest-param-test.h:350:10: note: in implicit copy constructor for 'testing::internal::ValueArray2<bool, bool>' first required here
  return internal::ValueArray2<T1, T2>(v1, v2);
         ^
/Users/karsten/workspace/ros2/ros2_master/src/ros/urdfdom/urdf_parser/test/gtest/include/gtest/gtest-param-test.h:1217:10: note: in instantiation of function template specialization 'testing::Values<bool, bool>' requested here
  return Values(false, true);
         ^
1 warning generated.
```

That makes a ROS2 master compilation with `-Werror` impossible on Clang.
Compiler details:
```
 ➭ clang++ --version
clang version 10.0.1
Target: x86_64-apple-darwin18.7.0
Thread model: posix
```

emersonknapp, 2021-02-16:
I'll also note that this warning is the only current build warning in the nightly clang+libcxx build

https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/698/clang/new/

clalancette, 2021-04-07:
This was fixed by #153, so closing.

closed 2021-04-07 by clalancette