ros/urdfdom #156: Branch urdfdom for Galactic

sloretz, 2021-04-22:
@clalancette @scpeters objections?

This one seems easy: create a `galactic` branch from the `ros2` branch and update rosdistro to say `galactic` is the devel branch.

scpeters, 2021-04-22:
sounds fine to me

clalancette, 2021-04-23:
Works for me.

traversaro, 2021-06-17:
Sorry to ask, but I was trying to reconstruct the history and plan for this repo (see https://github.com/ros/urdfdom/pull/146#issuecomment-722505443), and I wanted to ask if the plan is still to eventually merge all the ROS2-improvements back in master (as mentioned in https://github.com/ros/urdfdom/pull/146#issuecomment-722505443)? 
I would be happy to help, even if I know that also in that case the management effort on your side is not null. : )

clalancette, 2021-06-30:
@traversaro Sorry for the delay.  Yes, we definitely plan to merge the two branches back together.