ros/urdfdom #166: [cmake] urdf_LIBRARIES is still wrong

JackBoosY, 2022-01-06:
When the user uses multiple configurations (dbg/rel/rel_with_dbg), `urdf_LIBRARIES` should be a cmake expression containing these configuration information, but at this time, using one `find_library` can only search for a configured library.

Suggetsions:
```cmake
set(urdf_LIBRARIES urdfdom::urdfdom_model urdfdom::urdfdom_world urdfdom::urdfdom_sensor urdfdom::urdfdom_model_state)
```

traversaro, 2022-04-03:
There is already a variable called `urdfdom_TARGETS` (see https://github.com/conda-forge/urdfdom-feedstock/blob/4f5f046cb6f8c02d90e901356d17dc1052eb7e8c/recipe/test/CMakeLists.txt#L9 for an example of its use), I guess that one can be used by downstream libraries to link all the targets exported by urdfdom.

JackBoosY, 2022-04-06:
@traversaro Yeah, that should work. But it's just a workaround, not a fix.