ros/urdfdom #185: Merge branch of `scpeters/tinyxml2`

felixf4xu, 2023-05-19:
Hi,

I'm trying to compile ros2 from source on Windows 11 and `urdfdom` is the only repo that I need to change the code manually.

The reason is the dependency of "tinyxml" and I really hope "tinyxml2" should be used.

Reason:
- Tinyxml is not maintained for 10 years
- Tinyxml uses `catkin` which is not usually used in ROS2
- Tinyxml2 is another popular project and its main purpose is to replace the outdated tinyxml

When I check the code and issues, I found that the current branch of [scpeters/tinyxml2] (https://github.com/ros/urdfdom/tree/scpeters/tinyxml2) did exactly what I wanted and it also provide and cmake option to be back compatibe (which  I don't think it's necessary, though)

So, please merge the branch and say goodbay to tinyxml.

chging, 2023-10-25:
i have the same question, why not use tinyxml2

scpeters, 2023-12-06:
this will be fixed by https://github.com/ros/urdfdom/pull/186

closed 2023-12-06 by clalancette