ros/urdfdom #195: Add support for quaternion in URDF

doisyg, 2024-01-20:
Following conversation here: https://github.com/ros/urdfdom/pull/123#issuecomment-1865255612

Reopening as new PRs
- urdfdom: https://github.com/ros/urdfdom/pull/194
- urdfdom_headers: https://github.com/ros/urdfdom_headers/pull/77

And a draft on urdf_parser_py (because I never use python and have no proper clue of what's going on): https://github.com/ros/urdf_parser_py/pull/84

traversaro, 2024-01-20:
Thanks for proposing this! Given what was discussed in https://github.com/ros/urdfdom/pull/123#issuecomment-1865255612, I think that it is useful to clarify what version number this is new URDF+quaternion will use (as requested in https://github.com/ros/urdfdom/pull/123#issuecomment-1865255612). 

It is going to be URDF spec `2.0` or `1.1`? I am not an expert of versioning for file formats, and the usually used [semantic versioning](https://semver.org/) refers to software APIs (with corresponding ABIs), not file formats. The only example that I know of strictly defined versioning rules for a file format are FMI's versioning rules, defined in https://fmi-standard.org/docs/3.0/#VersioningLayered .

However, philosophy aside, the version support was added in https://github.com/ros/urdfdom/pull/133, but unless I am missing something no explicit semantics was defined for it. However, there is an implicit semantic defined by the check in https://github.com/ros/urdfdom/blob/29426ec119e9c3a9df9fc705d08c262a97b7f285/urdf_parser/src/model.cpp#L128 : all existing urdfdom version will refuse to load any document that URDF that have a version different from "1.0" . From that point of view, this means that we can safely call the URDF with quaternion both `1.1` or any other number, as old urdfdom that do not read quaternions will always do the right thing: refuse to load the document. 

Regardless of weather we use "1.1" or "2.0" as a version, I guess we need to modify the following point in the code:
* https://github.com/ros/urdfdom/blob/29426ec119e9c3a9df9fc705d08c262a97b7f285/urdf_parser/src/model.cpp#L128 : urdfdom should now accept the new version as well
* https://github.com/ros/urdfdom/blob/14f6db14859bfc5f430cf854bd4c77d2977126ab/urdf_parser/src/pose.cpp#L110 : Not sure about this, but perhaps quaternion should be parsed only if the urdf version is explicitly specified as 1.1/2.0, and print a clear error message otherwise? In this way we make sure that people that use quaternion properly set the version in their urdf 
* https://github.com/ros/urdfdom/pull/194/files#diff-f6e0c5c27bb1c6f9c99cab653d4849b0fe93a493cf60b5f0fcf2bf408588d7c6R4 : the updated spec is now the one of 1.1/2.0 URDF spec

traversaro, 2024-01-20:
> https://github.com/ros/sdformat_urdf ?

I guess no modification is required in this? This code creates a `urdf::ModelInterfaceSharedPtr`  from a SDF model, without passing via the textual representation of the URDF, that is modified by this proposal. In memory, the `urdf::Rotation` struct already uses quaternion.

> Whatever gazebo is using when doing gz sdf -p robot.urdf > robot.sdf or when spawning robot from an urdf robot description ? @ahcorde

This code is the URDF parser in https://github.com/gazebosim/sdformat/blob/sdf14/src/parser_urdf.cc . However, I do not think it needs any modification as it already uses urdfdom and so it reads the urdf pose from urdfdom's structures, see https://github.com/gazebosim/sdformat/blob/sdf14/src/parser_urdf.cc#L2802 . However, I think the problem here is of deployment. If I recall correctly (but things may have changed, I do not have a Humble install to quickly check) sdformat and gazebo packages use the urdfdom that ships with official Debian/Ubuntu repos (apt package: liburdfdom-dev, see for example https://packages.ubuntu.com/jammy/liburdfdom-dev), while ROS2 installs its own version of urdfdom package (ros-humble-urdfdom or something similar, see http://repo.ros2.org/ubuntu/main/pool/main/r/ros-humble-urdfdom/). So, to ensure that any change is available and supported in Gazebo for ROS Jazzy, we would need to make sure that urdfdom with this modifications is available in Ubuntu 24.04 repos. I am not sure if we are still in time for that, probably @j-rivero knows more.

traversaro, 2024-01-20:
Another related question: unfortunatly we still do not have the spec in a github repo (mainly my fault, see https://github.com/ros/urdfdom_headers/issues/62). However, could you propose here or somewhere how you would edit the spec in http://wiki.ros.org/urdf/XML to document quaternions?

ahcorde, 2025-11-24:
PRs are merged https://github.com/ros/urdfdom/pull/194 and https://github.com/ros/urdfdom_headers/pull/77

saikishor, 2025-11-24:
> fyi @mjcarroll @saikishor given that we discuss of this at ROSCon .

@traversaro how do you suggest to proceed?. We need to get the versioning work done properly.

scpeters, 2026-09-24:
quaternion support is in Lyrical in URDF 1.1 (see README)

closed 2026-09-24 by scpeters

doisyg, 2026-09-24:
And that's an absolutely amazing news ! Thanks to everybody who took it to the finish line