ros/urdfdom #199: support export URDF

briansoe66, 2024-03-14:
I would like to use the export functions. However, they are deprecated, so that the tinyxml2 dependency is private, as described in: [PR #190](https://github.com/ros/urdfdom/pull/190)   

I propose changing the API, so that export is available does not use tinyxml2. If this is acceptable, I can create a PR.

```
URDFDOM_DLLAPI std::string  exportURDF(ModelInterfaceSharedPtr &model);

URDFDOM_DLLAPI std::string  exportURDF(const ModelInterface &model);
```

clalancette, 2024-03-14:
In general, I am OK with that, but you are going to have to call them something different since we want to leave the existing deprecated ones in place for now .  That's because in C++ you can't have an overload that differs only in the return type.

RobertWilbrandt, 2025-04-17:
For the record: I am also currently using this API for a tool change system that builds a composite URDF at runtime depending on attached tools. I don't require access to tinyxml and would be fine with a string export.

I'd say something like `urdf::exportModel` would conceptually make sense: `parseUrdf` converts URDF to a model and this function should go the opposite direction. I'd be happy to create a PR with this in the near future.

chhtz, 2025-09-16:
This could also be something like `exportURDFtoStream`, with an additional `std::ostream&` argument (in case someone wants to write the output to a file, there is no need to store it in a string first -- and to generate a string, one could stream it into a `stringstream`).

N.B. It would have been nice not to remove deprecated functions, as long as there are pending issues regarding having an alternative ...

bsoe-flexiv, 2026-09-16:
This issue will be fixed by https://github.com/ros/urdfdom/pull/256